romilly/quick2wire-gpio-admin #5: gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio22/direction: No such file or directory

zmofei, 2015-02-11:
after install I can't find folder `gpio` in ` /sys/devices/virtual/`

greggpeacock, 2015-02-12:
Having the exact same problem! Hopefully we can sort that out.

greggpeacock, 2015-02-12:
https://github.com/raspberrypi/linux/issues/791

zmofei, 2015-02-12:
the path was change to another folder in new system?

lurch, 2015-02-12:
Don't mean to sound rude, but not sure why you're still relying on any of the quick2wire code, given that none of it has been touched in almost 2 years?
See also https://groups.google.com/forum/#!topic/quick2wire-users/OjJY9yweAws

If you do have to rely on such "frozen" code, then you should treat all the other software on your Pi as "frozen" too and not be doing software upgrades, because something is bound to break sooner or later... All IMHO of course.

aesopwolf, 2015-02-23:
Looks like there's a pull request to fix it https://github.com/quick2wire/quick2wire-gpio-admin/pull/6

ts00617355, 2015-03-07:
wow, #6 work for me, thx a lot.

note that if someone is using pi-gpio:
After changing the path and reinstalling gpio-admin, you need to change the path variable to ( sysFsPath = "/sys/class/gpio") in pi-gpio.js: line7 in node_modules/pi-gpio folder.

michaelderemer, 2015-05-15:
@ts00617355 I am using nodejs pi-gpio and can now successfully read pin 16, but i can't read any other pins. in my /sys/class/gpio/ i only have gpio23 -> ../ ../devices/soc/3df20000
How do i access other pins? It seems like they would each need their own directory. I'm not sure how i got gpio23's there. Any ideas?

lurch, 2015-05-16:
@michaelderemer Does https://www.kernel.org/doc/Documentation/gpio/sysfs.txt help?

Alternatively, take a look at http://raspi.tv/rpi-gpio

knocte, 2015-10-19:
@aesopwolf there's a better fix here: https://github.com/quick2wire/quick2wire-gpio-admin/pull/7

mvtcode, 2016-01-06:
help me:
Error when trying to open pin 16
gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio23/direction: No such file or directory

closed 2018-07-12 by zmofei