romilly/quick2wire-gpio-admin #10: /sys/devices/virtual/gpio doesn't exist

aramperez, 2015-06-17:
I'm using 2015-05-05-raspbian-wheezy on a Raspberry Pi 2 and when I run gpio-admin, I get the following error: "gpio-admin: failed to change group ownership of /sys/devices/virtual/gpio/gpio22/direction: No such file or directory". Does the program have to use the "/sys/devices/virtual/gpio..." folders or can it use the "sys/class/gpio..." folders?

pvgennip, 2015-09-24:
I have the same problem. It seems that the path to gpio has changed from /sys/devices/virtual/gpio to /sys/class/gpio. Using this in stead should solve your problem.

aramperez, 2015-09-24:
Hi Pvgennip, thanks for the info, I ended up using wiringPi.

knocte, 2015-10-19:
There's a fix for this: https://github.com/quick2wire/quick2wire-gpio-admin/pull/7

bkbonner, 2016-05-23:
+quickwire  Is there any reason #7 hasn't been merged yet?

leakingtapan, 2016-06-03:
@pvgennip could you give more inputs on how to change to use the new path? Thanks

pvgennip, 2016-06-04:
@leakingtapan check the changed file in #7, you can see how to use the correct path. Good luck!