romilly/quick2wire-gpio-admin #14: Getting this error please help!

Wouterkaas, 2019-08-13:
I get this error
Error when trying to open pin 11
gpio-admin: could not flush data to /sys/class/gpio/export: Device or resource busy
 Raspberry pi 3b+
config.json:
`      
  {
    "bridge": {
      "name": "Homebridge3",
      "username": "DC:22:3D:E3:CE:30",
      "port": 4000,
      "pin": "333-33-333"
    },
    
    "description": "This is an example configuration file with one fake accessory and one fake platform. You can use this as a template for creating your own configuration file containing devices you actually own.",
    "ports": {
      "start": 52100,
      "end": 52150,
      "comment": "This section is used to control the range of ports that separate accessory (like camera or television) should be bind to."
    },
    "accessories": [
      {
        "accessory": "relay",
        "name": "Relay 1",
        "pin" : 11,
        "duration" : 0
      },
      {
        "accessory": "relay",
        "name": "Relay 2",
        "pin" : 12,
        "duration" : 0
      }
    ],

      
        "platform": "CameraUI",
        "videoProcessor": "ffmpeg",
        "cameras": [
          {
            "name": "Flur",
            "active": true,
            "videoConfig": {
              "source": "-f v4l2 -r 30 -s 1280x720 -i /dev/video0",
              "maxWidth": 1920,
              "maxHeight": 1080,
              "maxFPS": 30
            }
          }
        ]
      
  }`

marcgrc, 2020-03-22:
I get this error
Error when trying to open pin 11
gpio-admin: could not flush data to /sys/class/gpio/export: Device or resource busy

clrmsc, 2020-07-08:
PLS HELP
Error when trying to open pin 7
gpio-admin: could not flush data to /sys/class/gpio/export: Device or resource busy